printers-for-people/catboat #30: support for lcd display(s)?

gurkburk76, 2023-08-07:
So i was thinking of giving my new anycubic kobra a little boost with kliper and found this project that *seems* to do it and also dosen't need me to go all crazy with the soldering iron, but what i was wondering if the lcd display of the printer will be good for anything after flahing ? 
I found some references to "lcd" when looking at issues and patches but hard to say if those atm actually brings back life to the display. not a huge thing but i'm curious.

Jookia, 2023-08-07:
There is currently no support for the LCD, adding it would require somehow integrating https://github.com/Desuuuu/klipper and https://github.com/Desuuuu/DGUS-reloaded-Klipper . Instead the way to use this project is remotely using Moonraker and a frontend like Mainsail.

gurkburk76, 2023-08-07:
No such port so i guess i have an old rev. MAny thanks, i'll try it out later anyway, seems to be better than marlin.

Jookia, 2023-08-07:
It can be better if you enjoy configuring printers and following online guides. :) Klipper is a bit of a rabbit hole in itself, and most Klipper users don't use a printer display simply because the Mainsail interface is so much better. Marlin is more designed for basic printing tasks from what I can see.

closed 2023-08-07 by Jookia