printers-for-people/catboat #31: Investigate clocking issue

Jookia, 2023-08-07:
I've been talking with @Fragmon a bit about their Kobra work today and an interesting find came up: Bltouch probes don't work with the clock speed set to 200mhz. Flashing a 200mhz config shows a 168mhz clock in Mainsail, flashing a 168mhz config fixes it.

This is confusing to me because I've always run at 200mhz and found 168mhz actually breaks on Trigorilla v1.0.4 boards.

However it seems the mainline Kobra source runs at 168mhz:

```
===================================================== ===============================

V2.8.2_20220905T1717

Firmware
1. Before clock initialization, turn on high performance mode
2. Fixed the problem that the minimum speed is limited to 5mms

UI
1. None

===================================================== ===============================

V2.8.1_20220725T1432

Firmware
1. Reduce the system frequency from 200M to 168M, compatible with 210807N batch
2. Use SRAM1 and SRAM2 instead of SRAMH
3. Optimize the problem of continuous printing after power failure without power failure during normal printing
```

I will have to devise some tests to get to the root of this.

Jookia, 2023-08-07:
Fragmon has confirmed this is on a Trigorilla A V1.0.4 board. After some quick inspection I didn't find much code in Klipper for setting the system clock, instead defering to the bootloader- I wonder if these chips have a default clock speed that Klipper isn't aware of. Maybe a chip revision or fuse setting?

Fragmon, 2023-08-08:
Datasheet: http://www.huazhoucn.com/upFiles/common/2023/04/HC32F460%20series%20Datasheet%20Rev1.3.pdf.pdf.pdf

Page: 58

Maybe anycubic by cheaper batch that only work with 168Mhz.

Jookia, 2023-09-12:
I touched on this yesterday and learned that the default clocks are set up by the bootloader and the current code assumes these are sane and make sense. I'm not so sure about this approach so I copied the Marlin code and shoved it in to a test branch here: https://github.com/printers-for-people/catboat/commit/017203002358a5f6161da017673f47de9715548f With this I have been able to underclock my printer to 168mhz from 200mhz successfully.

I have sent this out to someone having some printer issues and I'll see if this helps fix their weird issues. Either way I'm considering tidying up and including this just because setting the system clocks before using them is the right thing to do.

brillianthugo, 2023-09-29:
mcu(HC32F460)
Version: 7a2bc2af
Load: 0.00, Awake: 0.00, Freq: 168 MHz,

i new to klipper and using catboat , the system load is 168mhz , should i reflash it with 168mhz firmware?

Jookia, 2023-09-29:
Yes, reflash with 168mhz firmware.

Jookia, 2023-10-20:
I'm going to solve this by adding a note to try both clocking configurations. Without knowing why they underclocked these boards I don't feel comfortable suggesting people use clocks that aren't default.

leebobleebob, 2024-08-25:
Hi all exact same issue here I am pretty new to this so I have a working config with the 200mhz however I keep getting the error and stopping my print midway, so this is what I am trying:

Trigorilla A V1.0.4 board:

1. I set my Klipper Firmware Configuration as follows:

Micro-controller Architecture (Huada Semiconductor HC32F460)  --->
    Communication interface (Serial (PA3 & PA2))  --->
    Application Address (0x008000)  --->
    Clock Speed (168 MHz)  --->
    
2. Build out Firmware Output as follows:

###### Building firmware ...
  Creating symbolic link out/board
Loaded configuration '/home/admin/klipper/.config'
No changes to save (for '/home/admin/klipper/.config')
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/hc32f460/main.o
  Compiling out/src/hc32f460/interrupts.o
  Compiling out/src/hc32f460/gpio.o
  Compiling out/src/../lib/hc32f460/mcu/common/system_hc32f460.o
  Compiling out/src/../lib/hc32f460/driver/src/hc32f460_clk.o
  Compiling out/src/../lib/hc32f460/driver/src/hc32f460_efm.o
  Compiling out/src/../lib/hc32f460/driver/src/hc32f460_sram.o
  Compiling out/src/../lib/hc32f460/driver/src/hc32f460_utility.o
  Compiling out/src/../lib/hc32f460/driver/src/hc32f460_gpio.o
  Compiling out/src/../lib/hc32f460/driver/src/hc32f460_pwc.o
  Compiling out/src/hc32f460/adc.o
  Compiling out/src/../lib/hc32f460/driver/src/hc32f460_adc.o
  Compiling out/src/hc32f460/serial.o
  Compiling out/src/generic/serial_irq.o
  Compiling out/src/../lib/hc32f460/driver/src/hc32f460_usart.o
  Compiling out/src/hc32f460/hard_pwm.o
  Compiling out/src/../lib/hc32f460/driver/src/hc32f460_timera.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/generic/crc16_ccitt.o
  Building out/compile_time_request.o
Version: b25ed634
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
  Creating bin file out/klipper.bin
[✓ OK] Firmware built!

3. Flash output as follows:

/=======================================================\
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
[... 192 lines cut ...]
|  40) creality-v4.2.7                                  |
|  41) fly-gemini-v2                                    |
|  42) flyboard-mini                                    |
|  43) fysetc-cheetah                                   |
|  44) fysetc-cheetah-v2                                |
|  45) fysetc-s6-v1.2                                   |
|  46) fysetc-s6-v2                                     |
|  47) fysetc-spider                                    |
|  48) fysetc-spider-v1                                 |
|  49) generic-lpc1768                                  |
|  50) generic-lpc1769                                  |
|  51) mks-robin-e3                                     |
|  52) mks-robin-e3d                                    |
|  53) mks-sbase                                        |
|  54) mks-sgenl-v1                                     |
|  55) mks-sgenl-v2                                     |
|  56) monster8                                         |
|  57) robin_v3                                         |
|  58) smoothieboard-v1                                 |
|-------------------------------------------------------|
|                        Q) Quit                        |
\=======================================================/
###### Please select board type: 

Not sure which of these I should choose any help would be greatly appreciated :)

leebobleebob, 2024-08-25:
In casee this helps here is the " mostly working" firmware I am currently using.

[firmware.bin.zip](https://github.com/user-attachments/files/16740381/firmware.bin.zip)

Jookia, 2024-08-26:
Hi there, how did you originally flash that firmware? You should flash a new version the same way. After building it with KIAUH, find the klipper.bin file and upload it to an SD card as 'firmware.bin' then insert it in to the printer. This is like with the Ender 3 if that helps. It may be in ~/klipper/out/klipper.bin

leebobleebob, 2024-08-26:
I originally snagged the firmware from a guy that commented on a different issue, because I was hitting the same issue when trying it before, but it seems I was overthinking this, feeling like I had to use the flash menu in KIAUH to create that file, however there it was right in the klipper/out dir I did the following:

1. cd klipper/out
2. cp firmware.bin klipper/docs/klipper.bin
3. In the mainsail interface navigated to Machine >> (Under Config Files) Root (dropdown) changed to docs, found the klipper.bin file, hit the checkmark next to the file and then the download icon (upper right) that appeared after. (Avoided an scp process)
4. In my Downloads folder I got a new folder called docs, inside that folder there was a file called klipper.bin I made sure to reformat my sd card, changed the name of klipper.bin to firmware.bin copied the firmware.bin file to the sd card.
5. Made sure the power was off on the printer and the sd card inserted
6. Turned printer on, waited the suggested 5 min
7. Firmware rebooted the Mainsail interface and all seems to be good

Renamed it firmware.bin, moved it over to the sd, addded it to the printer waited for the firmware to take and have not seen the error since. 

I have one long print to do tonight so all will be revealed in the AM. 

Thanks so much for pointing me in the right direction! :)

Jookia, 2024-08-26:
Thanks for asking so I could help :)
Tell me how the print goes!

leebobleebob, 2024-08-26:
So I am wondering if I made the firmware bit incorrectly, everything works and then just into whatever project it may be, it absolutely craps the bed and I have to do a firmware restart. Uploaded my Klippy logs to see if that may help. I even completely started my Pi instance over again from scratch, wiped the sd card and built it all again and it seems like that part is stable but that new firmware is an issue. I even re-create the klipper.bin/firmware.bin again with the new instance just to be sure. Any help would be greatly appreciated.

[klippy-8.log.zip](https://github.com/user-attachments/files/16752584/klippy-8.log.zip)

Jookia, 2024-08-26:
Hello again,

This is probably it's own issue, could you open a separate GitHub issue for it?

That said, the logs say this:

```
b'Got EOF when reading from device'
Transition to shutdown state: Lost communication with MCU 'mcu'
```

This usually means a poor USB cable, even if it was the one that came with the
Kobra (it wasn't designed for Klipper use.)

Could you find a better cable and try it?

Jookia.

Jookia, 2024-11-04:
Okay, this issue has been solved for a while. Closing.

closed 2024-11-04 by Jookia