printers-for-people/catboat #33: Error Compiling Firmware for the Anycubic Kobra (Standard)

eriksonssilva, 2023-12-12:
Hello everyone! Hope you're doing well!
Thanks for (trying to) making our lives easier!
I have been trying to compile the firmware for the Kobra since yesterday, but no success.
I know that this may not be 100% related to Catboat, but I feel you guys would be able helping, since I've not found anything that would solve my particular problem.
I have successfully installed Catboat + Moonraker + Mainsail on a Linux Mint machine.
Reason: I don't have a raspPi here, but did have an old PC.
I can access mainsail's graphical interface, I can mess around with the files and all the stuff.
However, I cannot get the printer to communicate to it, because I don't have the klipper firmware on my Kobra.
I have tried (tried A LOT!) to find the already compiled firmware somewhere, but no avail (google, facebook, you name it).
So I thought it would be ok to try to compile my own firmware.
Long story short: It's not...
Through KIAUH, (after setting Catboat as the repository), I go to "Advanced", "Build Only". I set the Architecture to AVR and processor to Atmega328p (but also tried Atmega328 out of curiosity), and when I click "Q" and say "Yes", here's what I see:

`
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: region `text' overflowed by 4856 bytes
collect2: erro: ld returned 1 exit status
make: *** [Makefile:72: out/klipper.elf] Erro 1
`

I have tried flashing the image again, tried using Debian instead of Mint...

Tried all the suggestions found [here](https://klipper.discourse.group/t/cannot-build-firmware-for-arduino-uno/7041), [here](https://klipper.discourse.group/t/issue-compiling-firmware-for-atmega328p/1028), [here](https://github.com/Klipper3d/klipper/issues/4938)...
I really don't know where to go from there... The simplest solution would be if anyone could send me the compiled firmware, but if someone has the solution for this it would be much appreciated.

Thanks a lot!

Jookia, 2023-12-12:
Hello there! It looks like you're close, but not quite there: The Kobra doesn't use an AVR or Atmega328p chip. It uses a HC32F460!

At the top of config/printer-anycubic-kobra-2022.cfg I have put this note:

```
# To use this config, during "make menuconfig" select the HC32F460 architecture
# and communication interface to serial on PA3 and PA2.
# Make sure the clock is set to 200MHz.

# Note that the "make flash" command does not work with this board.
# After running "make", rename the out/klipper.bin file to out/firmware.bin
# Copy the file out/firmware.bin to an SD card formatted to FAT32, with an
# MBR partition table. Then restart the printer with the SD card inserted.
```

Please tell me if this helps.

eriksonssilva, 2023-12-12:
Hello there @Jookia ! Thanks a million for your answer!
DOOH, I feel so dumb because I've read everything yet missed the most important part! haha
Ok, I was able compiling the firmware, YAY!
I have flashed the firmware (copied the file to a FAT32 formatted SD Card), and as soon as it powers on, I can hear 5 beeps.
Then I have the stationary "Anycubic" logo.
Powering it off/On, it remains on the anycubic logo screen...
Still, I am unable connecting to the printer:

[image]

[image]

And something I have noticed is that, before flashing the Klipper firmware, I was able to get the MCU on the KIAUH menu using the USB option.
Now, I'm only able getting it using the UART option...

I even changed the line:
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
to
serial: /dev/ttyS0 (since that's what KIAUH now returns to me)

but still no avail... I'm pretty sure this may be something simple, but I cannot wrap my head around it!

Thanks a lot for helping me making progress!

eriksonssilva, 2023-12-12:
GOT IT WORKING!!!

Ok, after a lot of trial and error, I found what was the issue: brltty
For some reason, this braille library gets ahold of the printer port for some reason I don't know...
I remember having removed this yesterday (before posting here) and that's the reason why I was able finding the MCU ID.

So after all the steps, I have removed brltty and now everything works!

For future reverence, if you have followed all the steps correctly but your printer isn't recognized, consider removing this by running

´sudo apt-get remove brltty´.

Thanks a million once again!

closed 2023-12-12 by eriksonssilva

Jookia, 2023-12-12:
Congratulations! Enjoy Klipper on your Kobra. :)