printers-for-people/catboat #36: Kobra Max Not Flashing (1.0.4 board; Resistor Not Moved)

Jahigs, 2024-02-14:
installed catboat via KIAUH, built firmware using the following params:
[image]
printer beeps 5 times when powered on, after letting it sit for a few minutes i powered down and brought the sd card back to my pc, the filename didn't change, doesn't appear it took my build. Any thoughts?

Jahigs, 2024-02-14:
Thanks for the quick response! So you were totally right, internet just had me down a rabbit hole with those beeps. I have klipper running currently and was able to issue a home command, but then received the following error:
[image]
any chance you have a working printer.cfg and macros.cfg for the max you could shoot my way?

Jookia, 2024-02-14:
Currently we don't have a configuration for the Kobra Max, but it should work if you base it on configs/printer-anycubic-kobra-2022.cfg

Jahigs, 2024-02-14:
I'll take any help I can get, i'm still trying to catch up when it comes to klipper. the current config, macros, and firmware.bin im using are attached in this thread, no rush obviously, but feel free to leave comments in the config on what you change so i can understand what went wrong if you figure this out before i do.
[Anycubic-Kobra-Max-Klipper-Conversion-main.zip](https://github.com/printers-for-people/catboat/files/14284551/Anycubic-Kobra-Max-Klipper-Conversion-main.zip)

Jookia, 2024-02-14:
Alright so to convert it for use with Catboat you have to change '[tmctrigorilla stepper_x]' to '[tmc2209 stepper_x]' and uart_address to 0 and remove the '[tmc2208 extruder]' section entirely. Both will use the stepper_x settings. If it gives more errors update here, I'm a bit rusty with this :)

Jahigs, 2024-02-14:
printer.cfg has been updated to:
`## Anycubic Kobra Max ##
## Compile for HC32F460 ##
## Serial PA3 & PA2 "Anycube"

[include mainsail.cfg]
[include macros.cfg]

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command
baud: 250000

### Enable this to use the Orange Pi as MCU.
### I haven't found anything useful for this yet as nothing works for SPI on it
#[mcu opi]
#serial: /tmp/klipper_host_mcu

### Enable this to use an RP2040 for your ADXL host. You cannot use the Orange Pi so this is the next best option.
#[mcu adxl]
# Change <mySerial> to whatever you found above. For example,
# usb-Klipper_rp2040_E661640843545B2E-if00
#serial: /dev/serial/by-id/usb-Klipper_rp2040_E660C062137B4E29-if00
#[adxl345]
#cs_pin: adxl:gpio1
#spi_bus: spi0a
#axes_map: x,z,y
#[resonance_tester]
#accel_chip: adxl345
#    203.5, 200, 20
#[output_pin power_mode] # Improve power stability
#pin: adxl:gpio23

[stepper_x]
step_pin: PA5
dir_pin: PA4
enable_pin: !PC3
microsteps: 64
rotation_distance: 40
endstop_pin: !PA6
position_min: -6
position_endstop: -5
position_max: 407
homing_speed: 100

[tmc2209 stepper_x]
uart_pin:PA15
tx_pin:PA9
#
diag_pin:PA6
sense_resistor: 0.100
run_current: 0.9
uart_address: 0
#
driver_SGTHRS: 0
interpolate: False
stealthchop_threshold: 999999

[stepper_y]
step_pin: PC4
dir_pin: PA7
enable_pin: !PC3
microsteps: 64
rotation_distance: 40
endstop_pin: !PC5
position_min: -23.3
position_endstop: -23.2
position_max: 400
homing_speed: 100

[tmc2209 stepper_y]
uart_pin:PA15
tx_pin:PA9
#diag_pin:^PC5
sense_resistor: 0.100
run_current: 1
uart_address: 1
#driver_SGTHRS: 50
interpolate: False
#stealthchop_threshold: 999999

### Don't forget to SET YOUR E-STEPS!
### Measure 120mm, then extrude 100mm
### G1 E100 F100
### step_distance = old_e_steps * ((120 - distance_to_mark) / 100)

[extruder]
max_extrude_only_distance: 100.0
step_pin: PC14
dir_pin: !PC15
enable_pin: !PC3
microsteps: 16
#rotation_distance: 7.71 #For stock extruder (To be calibrated !)
rotation_distance: 7.21
nozzle_diameter: 0.800
filament_diameter: 1.750
heater_pin: PA1
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC1
min_temp: 0
max_temp: 250
pressure_advance: 0.6

[stepper_z]
step_pin: PC7
dir_pin: !PC6
enable_pin: !PC3
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#endstop_pin: PA8
#position_endstop: 0
position_min: -20
position_max: 452
homing_speed: 5

[tmc2209 stepper_z]
uart_pin:PA15
tx_pin:PA9
#diag_pin:PA8
sense_resistor: 0.100
run_current: 0.9
uart_address: 2
#driver_SGTHRS: 0
stealthchop_threshold: 999999

[stepper_z1]
step_pin: PB1
dir_pin: !PB0
enable_pin: !PC3
microsteps: 16
rotation_distance: 8

[probe]
pin: !PB6
speed: 2.5
samples: 3
samples_tolerance_retries: 3
#z_offset: 0.05
activate_gcode: probe_reset
### This macro is needed to reset the factory strain gauge on the Kobra Max
[gcode_macro probe_reset]
gcode:
    SET_PIN PIN=probe_reset_pin VALUE=1
    G4 P500
    SET_PIN PIN=probe_reset_pin VALUE=0
    G4 P100

[output_pin probe_reset_pin]
pin: !PB7

[safe_z_home]
home_xy_position: 203.5, 200 # Center of bed
speed: 100
z_hop: 5               # Move up 5mm
z_hop_speed: 20
move_to_previous: False

[bed_mesh]
speed: 100
horizontal_move_z: 3
mesh_min: 19, 19
mesh_max: 380, 380
algorithm: bicubic
probe_count: 5, 5

[output_pin LED]
pin: mcu:PB8
pwm: False
value: 0

[heater_bed]
heater_pin: PA0
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC0
#control = pid
#pid_kp = 73.966
#pid_ki = 0.884
#pid_kd = 1547.747
min_temp: 0
max_temp: 130

[fan]
pin: PB9

[heater_fan hotend_fan]
pin: PA13
heater: extruder
heater_temp: 50.0

[heater_fan controller_fan]
pin: PA14
heater: heater_bed
heater_temp: 45.0

[filament_switch_sensor runout]
pause_on_runout: True
switch_pin: !PC13

[temperature_sensor pi_temp]
sensor_type: temperature_host
min_temp: 0
max_temp: 100

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 2500
max_z_velocity: 5
max_z_accel: 100

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 69.538
#*# pid_ki = 0.893
#*# pid_kd = 1353.388
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 28.130
#*# pid_ki = 1.875
#*# pid_kd = 105.487
#*#
#*# [probe]
#*# z_offset = -0.110
#*#
#*# [input_shaper]
#*# shaper_type_x = mzv
#*# shaper_freq_x = 68.2
#*# shaper_type_y = mzv
#*# shaper_freq_y = 33.4
`
[image]
this stopped the shutdown sequence at least :D

Jookia, 2024-02-14:
Close, but it looks like you haven't changed '[tmc2209 stepper_x]' to '[tmctrigorilla stepper_x]'. This is the custom driver that works around the hardware bug.

Jahigs, 2024-02-15:
Morning brotha! I misread your comment from before about that name, my apologies. we're homing finally! Any idea why the Y axis sounds so bad when motor is active or moving for the kobra max? current config:
## Anycubic Kobra Max ##
## Compile for HC32F460 ##
## Serial PA3 & PA2 "Anycube"

[include mainsail.cfg]
[include macros.cfg]

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command
baud: 250000

[stepper_x]
step_pin: PA5
dir_pin: PA4
enable_pin: !PC3
microsteps: 64
rotation_distance: 40
endstop_pin: !PA6
position_min: -6
position_endstop: -5
position_max: 407
homing_speed: 100

[tmctrigorilla stepper_x]
uart_pin:PA15
tx_pin:PA9

diag_pin:PA6
sense_resistor: 0.100
run_current: 0.9
uart_address: 0

driver_SGTHRS: 0
interpolate: False
stealthchop_threshold: 999999

[stepper_y]
step_pin: PC4
dir_pin: PA7
enable_pin: !PC3
microsteps: 64
rotation_distance: 40
endstop_pin: !PC5
position_min: -23.3
position_endstop: -23.2
position_max: 400
homing_speed: 100

[tmc2209 stepper_y]
uart_pin:PA15
tx_pin:PA9
#
diag_pin:^PC5
sense_resistor: 0.100
run_current: 1
uart_address: 1
#
driver_SGTHRS: 50
interpolate: False
#stealthchop_threshold: 999999

### Don't forget to SET YOUR E-STEPS!
### Measure 120mm, then extrude 100mm
### G1 E100 F100
### step_distance = old_e_steps * ((120 - distance_to_mark) / 100)

[extruder]
max_extrude_only_distance: 100.0
step_pin: PC14
dir_pin: !PC15
enable_pin: !PC3
microsteps: 64
#rotation_distance: 7.71 #For stock extruder (To be calibrated !)
rotation_distance: 7.21
nozzle_diameter: 0.800
filament_diameter: 1.750
heater_pin: PA1
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC1
min_temp: 0
max_temp: 250
pressure_advance: 0.6

[stepper_z]
step_pin: PC7
dir_pin: !PC6
enable_pin: !PC3
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#endstop_pin: PA8
#position_endstop: 0
position_min: -20
position_max: 452
homing_speed: 5

[tmc2209 stepper_z]
uart_pin:PA15
tx_pin:PA9
#diag_pin:PA8
sense_resistor: 0.100
run_current: 0.9
uart_address: 2
#driver_SGTHRS: 0
stealthchop_threshold: 999999

[stepper_z1]
step_pin: PB1
dir_pin: !PB0
enable_pin: !PC3
microsteps: 16
rotation_distance: 8

[probe]
pin: !PB6
speed: 2.5
samples: 3
samples_tolerance_retries: 3
#z_offset: 0.05
activate_gcode: probe_reset
### This macro is needed to reset the factory strain gauge on the Kobra Max
[gcode_macro probe_reset]
gcode:
    SET_PIN PIN=probe_reset_pin VALUE=1
    G4 P500
    SET_PIN PIN=probe_reset_pin VALUE=0
    G4 P100

[output_pin probe_reset_pin]
pin: !PB7

[safe_z_home]
home_xy_position: 203.5, 200 # Center of bed
speed: 100
z_hop: 5               # Move up 5mm
z_hop_speed: 20
move_to_previous: False

[bed_mesh]
speed: 100
horizontal_move_z: 3
mesh_min: 19, 19
mesh_max: 380, 380
algorithm: bicubic
probe_count: 5, 5

[output_pin LED]
pin: mcu:PB8
pwm: False
value: 0

[heater_bed]
heater_pin: PA0
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC0
control = pid
pid_kp = 73.966
pid_ki = 0.884
pid_kd = 1547.747
min_temp: 0
max_temp: 130

[fan]
pin: PB9

[heater_fan hotend_fan]
pin: PA13
heater: extruder
heater_temp: 50.0

[heater_fan controller_fan]
pin: PA14
heater: heater_bed
heater_temp: 45.0

[filament_switch_sensor runout]
pause_on_runout: True
switch_pin: !PC13

[temperature_sensor pi_temp]
sensor_type: temperature_host
min_temp: 0
max_temp: 100

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 2500
max_z_velocity: 5
max_z_accel: 100

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 69.538
#*# pid_ki = 0.893
#*# pid_kd = 1353.388
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 28.130
#*# pid_ki = 1.875
#*# pid_kd = 105.487
#*#
#*# [probe]
#*# z_offset = -0.110
#*#
#*# [input_shaper]
#*# shaper_type_x = mzv
#*# shaper_freq_x = 68.2
#*# shaper_type_y = mzv
#*# shaper_freq_y = 33.4

Jookia, 2024-02-15:
Glad it's homing :) I'm not too sure, maybe tweak stealthchop.

Jahigs, 2024-02-15:
I think that did it! Thanks for all the help man :) not sure if you want it but just incase you run into another goob like me in the future, this is the combo that got my stock 1.0.4 kobra max running:
[Kobra Max.zip](https://github.com/printers-for-people/catboat/files/14300700/Kobra.Max.zip)

Jookia, 2024-02-16:
Thanks for that, I'll maybe add a config when I get time!

Jookia, 2024-11-04:
I have started the process of merging this config in to Catboat. See #44

closed 2024-11-04 by Jookia