printers-for-people/catboat #38: BL Touch not working

dog2bert, 2024-02-21:
I try the following config on my catboat install on BTT PAD7 on a Kobra Max, with the MakerPanda board for Stealthburner.

[bltouch]
sensor_pin: !PB6
control_pin: PB7
pin_up_reports_not_triggered: False
#probe_with_touch_mode: True
#speed: 20.0
#samples: 3
#samples_tolerance_retries: 3
#stow_on_each_sample: false
Z_offset: 0

The down command will work, but up command makes in oscillate up and down constantly. 
Homing, G28, BL touch failed to deploy when it get to the z axis home part.

dog2bert, 2024-02-21:
[image]
[image]
[image]

https://www.aliexpress.us/item/3256804234851176.html?businessType=ProductDetail&srcSns=sns_Copy&spreadType=socialShare&bizType=ProductDetail&social_params=60343500115&aff_fcid=5c5d7111e4c94725b399d47e75fdd342-1706664703853-07549-_DFem7rj&tt=CPS_NORMAL&aff_fsk=_DFem7rj&aff_platform=portals-tool&sk=_DFem7rj&aff_trace_key=5c5d7111e4c94725b399d47e75fdd342-1706664703853-07549-_DFem7rj&terminal_id=4d4e7541cbce406fa44e7baaf76f6137&afSmartRedirect=y&gatewayAdapt=glo2usa4itemAdapt

Jookia, 2024-02-21:
How do you know PB6 and PB7 are the correct pins?

Fragmon, 2024-02-22:
Hey jookia. I help him with this Board (i am the creator and this Board work with this Pins). I never seen this Kind of error, that the bltouch goes up and down. @dog2bert can you Post the video with this pin movement?

Jookia, 2024-02-22:
Can you try testing the bltouch using https://www.klipper3d.org/BLTouch.html instructions?

Jookia, 2024-02-22:
> Here is the video: https://youtube.com/shorts/tg0fCxpEp0Q?feature=share

It looks like the probe is going up and down in its idle state?

dog2bert, 2024-02-22:
Yes, it will go up and down until I trigger it down. Then when I trigger it up, it goes back to up/down.

dog2bert, 2024-02-22:
Switching the wires and config it had the same behavior.

I added this line to the bltouch config:
pin_up_touch_mode_reports_triggered: false

and now it will go down during homing z axis, and stop when triggered, but gets this error after:

!! BLTouch failed to deploy

Jookia, 2024-02-23:
Hmm. I don't have a bltouch to test. Could you try the other Klipper options in the article I linked please?

Jookia, 2024-10-23:
Any update on this?

dog2bert, 2024-10-23:
I eventually got it to work, I think it was a combination of wiring and then klipper settings.

Jookia, 2024-10-23:
Thanks for the update, I'm glad it works.

closed 2024-10-23 by Jookia