printers-for-people/catboat #39: TMC 'stepper_z' reports error: GSTAT:      00000001 reset=1(Reset)

Jahigs, 2024-04-19:
Not sure what is causing this at the moment, I've had a file stop mid print twice now, but it stopped in two different areas it appears. I've also managed to get through this print 1 time in between these failures, which is odd. Klippy.log doesn't have any relevant info about it when i try to search for the line, but i'll attach it just in case. Any ideas?
[logs-20240419-092409.zip](https://github.com/printers-for-people/catboat/files/15041004/logs-20240419-092409.zip)

Jookia, 2024-04-19:
Hmm. This error means the TMC2209 chip has reset, which isn't good. My first thought is you're having a power spike of some sort. Not sure.

Jahigs, 2024-04-19:
Hmm, when you say power spike, are you referring to the machine itself or the pi? I tried to run the print again, just had it fail in another random spot. I actually did get some logging info that shows the message though:
TMC 'stepper_z' reports GSTAT:      00000001 reset=1(Reset)
Transition to shutdown state: TMC 'stepper_z' reports error: GSTAT:      00000001 reset=1(Reset)
Dumping gcode input 0 blocks
Dumping 20 requests for client 1964277552
Received 151930.549634: b'{"id": 1726549072, "method": "register_remote_method", "params": {"response_template": {"method": "shutdown_machine"}, "remote_method": "shutdown_machine"}}'
Received 151930.552672: b'{"id": 1726592752, "method": "register_remote_method", "params": {"response_template": {"method": "reboot_machine"}, "remote_method": "reboot_machine"}}'
Received 151930.555131: b'{"id": 1726592752, "method": "register_remote_method", "params": {"response_template": {"method": "pause_job_queue"}, "remote_method": "pause_job_queue"}}'
Received 151930.557488: b'{"id": 1726592752, "method": "register_remote_method", "params": {"response_template": {"method": "start_job_queue"}, "remote_method": "start_job_queue"}}'
Received 151930.564021: b'{"id": 1726594064, "method": "objects/query", "params": {"objects": {"heaters": null}}}'
Received 151930.564666: b'{"id": 1726594128, "method": "objects/query", "params": {"objects": {"heaters": null}}}'
Received 151930.599774: b'{"id": 1726593040, "method": "info", "params": {}}'
Received 151930.601819: b'{"id": 1726594224, "method": "objects/list", "params": {}}'
Received 151930.601819: b'{"id": 1726593424, "method": "gcode/help", "params": {}}'
Received 151930.631667: b'{"id": 1726557936, "method": "info", "params": {}}'
Received 151930.631667: b'{"id": 1726556752, "method": "objects/list", "params": {}}'
Received 151930.633510: b'{"id": 1726557328, "method": "gcode/help", "params": {}}'
Received 151930.644818: b'{"id": 1726594352, "method": "objects/subscribe", "params": {"objects": {"webhooks": null, "configfile": null, "mcu": null, "gcode_move": null, "print_stats": null, "virtual_sdcard": null, "pause_resume": null, "display_status": null, "gcode_macro CANCEL_PRINT": null, "gcode_macro PAUSE": null, "gcode_macro RESUME": null, "gcode_macro SET_PAUSE_NEXT_LAYER": null, "gcode_macro SET_PAUSE_AT_LAYER": null, "gcode_macro SET_PRINT_STATS_INFO": null, "gcode_macro _TOOLHEAD_PARK_PAUSE_CANCEL": null, "gcode_macro _CLIENT_EXTRUDE": null, "gcode_macro _CLIENT_RETRACT": null, "gcode_macro LOAD_FILAMENT": null, "gcode_macro UNLOAD_FILAMENT": null, "gcode_macro START_PRINT": null, "gcode_macro END_PRINT": null, "gcode_macro DO_A_MESH": null, "gcode_macro TEST_SPEED": null, "gcode_macro BED_MESH_CALIBRATE": null, "gcode_macro _KAMP_Settings": null, "exclude_object": null, "stepper_enable": null, "tmc2209 stepper_y": null, "tmc2209 stepper_z": null, "z_tilt": null, "probe": null, "gcode_macro probe_reset": null, "output_pin probe_reset_pin": null, "bed_mesh": null, "output_pin LED": null, "heaters": null, "heater_bed": null, "fan": null, "heater_fan hotend_fan": null, "heater_fan controller_fan": null, "filament_switch_sensor runout": null, "temperature_host pi_temp": null, "temperature_sensor pi_temp": null, "motion_report": null, "query_endstops": null, "idle_timeout": null, "system_stats": null, "manual_probe": null, "toolhead": null, "extruder": null}, "response_template": {"method": "process_status_update"}}}'
Received 151930.800410: b'{"id": 1745437904, "method": "objects/subscribe", "params": {"objects": {"webhooks": null, "configfile": null, "mcu": null, "gcode_move": null, "print_stats": null, "virtual_sdcard": null, "pause_resume": null, "display_status": null, "gcode_macro CANCEL_PRINT": null, "gcode_macro PAUSE": null, "gcode_macro RESUME": null, "gcode_macro SET_PAUSE_NEXT_LAYER": null, "gcode_macro SET_PAUSE_AT_LAYER": null, "gcode_macro SET_PRINT_STATS_INFO": null, "gcode_macro _TOOLHEAD_PARK_PAUSE_CANCEL": null, "gcode_macro _CLIENT_EXTRUDE": null, "gcode_macro _CLIENT_RETRACT": null, "gcode_macro LOAD_FILAMENT": null, "gcode_macro UNLOAD_FILAMENT": null, "gcode_macro START_PRINT": null, "gcode_macro END_PRINT": null, "gcode_macro DO_A_MESH": null, "gcode_macro TEST_SPEED": null, "gcode_macro BED_MESH_CALIBRATE": null, "gcode_macro _KAMP_Settings": null, "exclude_object": null, "stepper_enable": null, "tmc2209 stepper_y": null, "tmc2209 stepper_z": null, "z_tilt": null, "probe": null, "gcode_macro probe_reset": null, "output_pin probe_reset_pin": null, "bed_mesh": null, "output_pin LED": null, "heaters": null, "heater_bed": null, "fan": null, "heater_fan hotend_fan": null, "heater_fan controller_fan": null, "filament_switch_sensor runout": null, "temperature_host pi_temp": null, "temperature_sensor pi_temp": null, "motion_report": null, "query_endstops": null, "idle_timeout": null, "system_stats": null, "manual_probe": null, "toolhead": null, "extruder": null}, "response_template": {"method": "process_status_update"}}}'
Received 151931.052066: b'{"id": 1726545040, "method": "objects/subscribe", "params": {"objects": {"webhooks": null, "print_stats": null, "heater_bed": null, "temperature_sensor pi_temp": null, "extruder": null, "configfile": null, "mcu": null, "gcode_move": null, "virtual_sdcard": null, "pause_resume": null, "display_status": null, "gcode_macro CANCEL_PRINT": null, "gcode_macro PAUSE": null, "gcode_macro RESUME": null, "gcode_macro SET_PAUSE_NEXT_LAYER": null, "gcode_macro SET_PAUSE_AT_L [... cut ...]
Stats 157521.1: gcodein=0  mcu: mcu_awake=0.001 mcu_task_avg=0.000006 mcu_task_stddev=0.000005 bytes_write=2833472 bytes_read=1407969 bytes_retransmit=120 bytes_invalid=839 send_seq=78075 receive_seq=78075 retransmit_seq=74596 srtt=0.004 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=199997092  heater_bed: target=65 temp=54.6 pwm=1.000 pi_temp: temp=53.7 sysload=0.19 cputime=3496.155 memavail=786948 print_time=5431.303 buffer_time=0.000 print_stall=0 extruder: target=220 temp=83.6 pwm=1.000

Jookia, 2024-04-19:
I mean the printer power but checking the Pi power could be good too.

Jahigs, 2024-04-25:
Update:
No power spikes on either the Pi or the Printer. After talking with others we've tried the following with no success on getting this error to disappear for more than 1 or 2 prints:
-Added a lot more cooling for the printer controller board
-Dropped run current for tmc2209 stepper_z from 0.9 to 0.6

Open to any suggestions on what could be going wrong here. I was reading another post between you and mykepredko about uart communications for z and x being shared which has me wondering if this could be related to this issue. Last thought is part failure; but is it more likely the motor failed, the tmc driver chip failed, or something else possibly?

Jookia, 2024-04-25:
It looks like this check is done every second by Klipper.
Marlin might not do these and mask the issue.

The communications between E and X are shared, not Z and X.
If anything the lack of UART reading from E and X could be masking errors on
those chips.

After this error happens could you dump the TMC registers for me?

    DUMP_TMC STEPPER=stepper_y
    DUMP_TMC STEPPER=stepper_z

Here's the TMC2209 datasheet:
https://www.analog.com/media/en/technical-documentation/data-sheets/TMC2209_datasheet_rev1.09.pdf

It says:

> To reset the chip to power on defaults, any of the supply voltages monitored
> by internal reset circuitry (VS, +5VOUT or VCC_IO) must be cycled. As +5VOUT
> is the output of the internal voltage regulator, it cannot be cycled via an
> external source except by cycling VS. It is easiest and safest to cycle
> VCC_IO to completely reset the chip.

Reset apparently requires Vs to drop to 4.5, V5vout to 3.5, or Vvcc_io to 2.55v.
That's a pretty nasty condition and I would expect it to affect other parts of the board too.

At this point I would try and create a stress situation to find what provokes the issue.
Maybe something like STEPPER_BUZZ would do it?
This can help narrow down to if it's ONLY stepper_z that has reset. It may not be the source
of the problem, something else may be causing a reset.

I see your stepper_x, stepper_y and stepper_e run_currents are also pretty
high. Could you drop them to 0.6?

Here's a long thread of things to check:

https://klipper.discourse.group/t/tmc-extruder-reports-error-gstat-00000001-reset-1-reset/1156/64

Please throughly inspect the wiring for your stepper motors if you can. You
don't want to find that this is due to some intermittent short that could lead
to a fire if it stops being intermittent. :)

Jookia, 2024-04-29:
Looks like the stepper_y isn't being reset. I would investigate the fan too, but I'm not too sure how to test which component has failed here. Maybe you could replace the Z stepper and see if that works? Or do a large 2D space-filling fractal first layer that stresses X and Y but not Z? Then try doing it again but with Z and see if it makes the problem worse? Testing the fans would be good too.

Jahigs, 2024-05-01:
when you say replace the stepper do you mean the driver or the motor? Thinking out loud about the fan: daughter board could potentially have gone bad, or possibly the connector from daughter board to motherboard, but im not sure if i can test those ideas without having spare parts. looking into a 2d print that would stress x and y and i'll be printing it today. is there anything in specific i should be looking for in logs or anything in general you recommend to be checking for if/when the print fails? this is my first round with trying to identify and fix a problem like this. thanks again for your help <3

Jookia, 2024-05-02:
I mean the stepper motor itself. I don't think the driver is replacable, it is soldered on the board.
I'm not sure if the fan 'daughter board' could have gone bad- the PCB toolhead is very very simple.
Still worth checking though.

I'm honestly not sure how to test this issue aside from stressing the steppers.
At this point I think it's down to testing whether the board is busted or the
physical Z stepper is busted.

This is my first time troubleshooting this too. Please keep me updated.

kozross, 2024-05-02:
> I mean the stepper motor itself. I don't think the driver is replacable, it is soldered on the board. I'm not sure if the fan 'daughter board' could have gone bad- the PCB toolhead is very very [...]

Does the toolhead PCB have its own MCU? If so, a failure there is very possible.

Jookia, 2024-05-02:
No, there's no MCU.

Jookia, 2024-10-23:
Any update on this?