printers-for-people/catboat #42: Add support for Anycubic ACE Pro

Kaympe20, 2024-07-06:


Jookia, 2024-07-06:
Hi, could you explain more about what you want from this issue?

kozross, 2024-07-06:
This would require supporting the Kobra 3 first, which would require a Klipper port and config to its board.

Kaympe20, 2024-07-09:
@Jookia If the ACE Pro becomes supported in a universal firmware such as catboat, it could theoretically be used with other printers like the Voron 2.4 or Ender 3.

Jookia, 2024-07-09:
Thanks for clarifying, and thanks for caring to open this issue. :)

I looked more at the ACE Pro in a video review, and it looks like a filament changer that is controllable by the printer, and could indeed be used by other printers (including my sad Kobra).

The only way this might happen is if someone else does it, or if someone with a Kobra 3 and ACE Pro has time to work with me on reverse engineering it. I don't have any estimates on this given how new it is. The Kobra 3 might have a big block of Python code we can use to learn how the ACE Pro works and do a clean room implementation, OR it might be locked to a toolhead PCB running firmware we can't dump using an encrypted communication protocol.

Another thing worth noting is that Anycubic have done their best to lock down the Kobra 3 firmware so it's entirely possible that they've done the same with this product. This means we can't even run Catboat on it. So supporting this would only fit the use case of someone not using the Kobra 3 with the ACE Pro.

The biggest blocker here is whether someone with an ACE Pro wants to go down this path with me. They don't have to be particularly skilled: I'm happy to provide instructions and things to test and try. But it will take a while and may not be doable. The main currency here is time, interest and patience.

Jookia, 2024-07-22:
Slight update on my above post: The locked down firmware is the Kobra 2 but I expect the same situation for the Kobra 3. However some reverse engineering has been done for the Kobra 2, so it's possible we can use that to dump things for the Kobra 3.

Curiosity got the best of me and I had a look online for more information about the ACE Pro and found these resources:

- Anycubic PTC troubleshooting: https://wiki.anycubic.com/en/error-codes/11530-code
- Anycubic timeout troubleshooting: https://wiki.anycubic.com/en/error-codes/11513-code
- Anycubic idler replacement guide: https://wiki.anycubic.com/en/fdm-3d-printer/kobra-3-combo/ace-pro-idler-pulley-assembly-replacement-guide
- Anycubic motor replacement guide: https://wiki.anycubic.com/en/fdm-3d-printer/kobra-3-combo/ace-pro-motor-replacement-guide
- Anycubic mainboard replacement guide: https://wiki.anycubic.com/en/fdm-3d-printer/kobra-3-combo/ace-pro-mainboard-replacement-guide
- Rookie 3D Prints' disassembly: https://youtu.be/uM8r_jgk06Y
- Scratch 3DPrinting's disassembly: https://youtu.be/Og_sR1fzotA

Externally, there's a mains connector, a 4 pin cable, and a 6 pin cable, but only the 6 pin cable seems to be used in practice, despite being connected to the printer as a 4 pin cable. It has four bowden tubes out and relies on the toolhead to cut and select which filament tube to use.  It seems like this could be controlled with Klipper and used on another printer using some kind of Y merging device, but that's kind of a rough sell given there's better solutions out there.

My guess as to what's happening here is the ACE is running Klipper and talking over CAN bus to the Kobra 3. The reason it has a 6 pin connector is probably related to terminating the CAN bus with a 120 resistor, the final printer would use a cable including the terminating resistor while chaining would use a different cable without the resistor.

Jookia, 2024-07-22:
I've opened a discussion at https://github.com/Bushmills/Anycubic-Kobra-3-rooted/discussions/2 to see if someone wants to provide information about the Kobra 3's interaction with the ACE Pro.

avxmw, 2024-08-24:
> I've opened a discussion at [Bushmills/Anycubic-Kobra-3-rooted#2](https://github.com/Bushmills/Anycubic-Kobra-3-rooted/discussions/2) to see if someone wants to provide information about the Kobra 3's interaction with the ACE Pro.

Hello,

I've got a Kobra 3 Combo with root access via adb/ssh as I did not update the printer itself to 2.3.3.1+ (which disables ssh and probably also adb), only updated the ACE. Current versions on my machine are 2.3.2.3 for the printer, 1.3.80 for the ACE.

As long as risk of permanent damage is low or zero, I'm willing to be guided. I know my way around Linux systems, however hardware or low level protocols are not my thing.

I'm in Germany and willing to hop on IRC somewhere (pls no discord) or do it via mail contact.

Jookia, 2024-08-24:
Hi there! Thanks for the offer! Risk should be minimal to none I imagine.

I'm Jookia on Libera or OFTC IRC if you want to ping me, but you can also email me at contact@jookia.org . I also have Matrix if that helps @Jookia:matrix.org .

wiltwong, 2024-08-26:
> I've opened a discussion at [Bushmills/Anycubic-Kobra-3-rooted#2](https://github.com/Bushmills/Anycubic-Kobra-3-rooted/discussions/2) to see if someone wants to provide information about the Kobra 3's interaction with the ACE Pro.

I added instructions on how to capture the serial traffic.

Jookia, 2024-09-01:
I've started working on this in another repository, so I'm going to close this issue *for now*. Eventually catboat will have to integrate support for this somehow, but I don't want communication fragmented. https://github.com/printers-for-people/ACEResearch

closed 2024-09-01 by Jookia

Jookia, 2025-02-19:
Just as an update, there is now a klipper plugin for this: https://github.com/utkabobr/DuckACE

agrloki, 2025-07-29:
> Just as an update, there is now a klipper plugin for this: https://github.com/utkabobr/DuckACE

It is freezed now :(
But work in progress on this:
https://github.com/agrloki/ValgACE/tree/main
https://github.com/BlackFrogKok/BunnyACE/tree/development