printers-for-people/catboat #47: Level sensor inconsistency triggers on Kobra Standard

mminella, 2024-11-13:
I have an original Kobra standard. I have upgraded it to a dual z stepper arrangement about a year ago and that was working fine with the original firmware. I am now trying to setup Klipper (using the Catboat fork) and am running into issues related to the Z axis. I have tuned the X and Y motors so they home fine, but I am running into two issues right now with the Z:

- Inconsistent homing on the Z axis - When I home the extruder, X and Y perform as expected. Sometimes the Z goes down until the leveling sensor trips (indicated by the red light being on) but some times it does not and just sits above the bed. I have also had one or two "Endstop z still triggered after retract" errors as well during homing.

- Inconsistent errors when probing during bed leveling - Seemingly randomly, I'll get "Probe triggered prior to movement" errors during probes on bed leveling. It's important to note that these do not always happen on the first probe of a position so I don't think it's related to the hight of the moves in between probe positions.

I used this method (https://www.klipper3d.org/TMC_Drivers.html#tuning) to tune the X and Y stepper motors but the Z ones basically ignore that (and just goes) which I think is the correct behavior given the leveling sensor. For the record, below is my printer.cfg. What am I missing?

```
[include mainsail.cfg]
[virtual_sdcard]
path: /home/michaelminella/printer_data/gcodes
on_error_gcode: CANCEL_PRINT

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method: command

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 2000
max_z_velocity: 5
max_z_accel: 100

# Steppers

[stepper_x]
step_pin: PA5
dir_pin: PA4
enable_pin: !PC3
rotation_distance: 40
microsteps: 16
endstop_pin: tmctrigorilla_stepper_x:virtual_endstop
position_min: -42
position_endstop: -42
position_max: 224
homing_speed: 50 
homing_retract_dist: 0

[tmctrigorilla stepper_x]
uart_pin: PA15
run_current: 0.8
stealthchop_threshold: 999999
uart_address: 0
driver_SGTHRS: 115
diag_pin: ^PA6

[stepper_y]
step_pin: PC4
dir_pin: PA7
enable_pin: !PC3
rotation_distance: 40
microsteps: 16
endstop_pin: tmc2209_stepper_y:virtual_endstop
position_min: -3
position_endstop: -3
position_max: 222
homing_speed: 50
homing_retract_dist: 0

[tmc2209 stepper_y]
uart_pin: PA15
run_current: 0.8
stealthchop_threshold: 999999
uart_address: 1
driver_SGTHRS: 120
diag_pin: ^PC5

[stepper_z]
step_pin: PC7
dir_pin: !PC6
enable_pin: !PC3
rotation_distance: 4
microsteps: 16
endstop_pin: probe:z_virtual_endstop
position_min: -1
position_max: 250
homing_speed: 10

[tmc2209 stepper_z]
uart_pin: PA15
run_current: 0.8
stealthchop_threshold: 999999
uart_address: 2
driver_SGTHRS: 0
diag_pin: ^PA8

[extruder]
step_pin: PC14
dir_pin: PC15
enable_pin: !PC3
microsteps: 16
rotation_distance: 7.794750
nozzle_diameter: 0.400
filament_diameter: 1.750
max_extrude_only_distance: 105
pressure_advance: 0.045
heater_pin: PA1
sensor_type: ATC Semitec 104GT-2
sensor_pin: PC1
control: pid
# Calibrated with hot end at 200c, bed at 60, parts fan on
pid_Kp: 17.111
pid_Ki: 0.659
pid_Kd: 111.06
min_temp: 5
max_temp: 260

[heater_bed]
heater_pin: PA0
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC0
control: pid
# Calibrated with hot end at 200c, bed at 60, parts fan on
pid_Kp: 61.847
pid_Ki: 0.902
pid_Kd: 1059.902
min_temp: 5
max_temp: 110

[bed_mesh]
speed: 50
mesh_min: 10, 10
mesh_max: 210, 210
probe_count: 5, 5
horizontal_move_z: 7

[probe]
pin: PB8
x_offset: 42
y_offset: 3
z_offset = 0.230
samples: 3
samples_tolerance_retries: 2
speed: 3

# Cooling

[fan]
pin: PB9

[heater_fan extruder_fan]
pin: PA13

[controller_fan controller_fan]
pin: PA14
fan_speed: 0.4

# Extra

[filament_switch_sensor my_sensor]
switch_pin: ^!PC13

[safe_z_home]
home_xy_position: 110, 110

[output_pin beeper]
pin: PB5
pwm: True
value: 0
shutdown_value: 0
cycle_time: 0.01
scale: 1

[gcode_macro M300]
gcode:
    # Use a default 1kHz tone if S is omitted.
    {% set S = params.S|default(1000)|int %}
    # Use a 10ms duration is P is omitted.
    {% set P = params.P|default(100)|int %}
    SET_PIN PIN=beeper VALUE=0.5 CYCLE_TIME={ 1.0/S if S > 0 else 1 }
    G4 P{P}
    SET_PIN PIN=beeper VALUE=0

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	-0.320417, -0.295833, -0.330417, -0.353750, -0.348750
#*# 	-0.194167, -0.140000, -0.166667, -0.168333, -0.178750
#*# 	-0.104167, -0.021250, -0.026667, -0.023750, -0.023333
#*# 	-0.101667, -0.017083, -0.007917, 0.008750, 0.015000
#*# 	-0.175000, -0.049167, -0.002083, 0.051250, 0.101667
#*# x_count = 5
#*# y_count = 5
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*# min_x = 10.0
#*# max_x = 210.0
#*# min_y = 10.0
#*# max_y = 210.0
#*#
#*# [probe]
#*# z_offset = 0.230
```

Jookia, 2024-11-13:
Hi there, thanks for opening this issue! :)

Could you add 'sample_retract_dist: 5.0' in to your [probe] section? My hunch is that your printer isn't lifting up high enough before probing. I had this problem on my Kobra so I had to add that.

If that resolves the issue please let me know.

Jookia, 2024-11-14:
On the Kobra both the Z endstop and probe are the same thing, so ultimately what's happening here is that the probe is triggering when it shouldn't. The sample_retract_dist should make the toolhead move up enough that the probe light turns off, can you tell me if that's happening?

A bit of an easier way to test this is to run the PROBE_ACCURACY g-code command after homing, if it works then please move the toolhead to somewhere else on the bed where it fails and run it again.

mminella, 2024-11-14:
The probe light does turn off when it is high enough. The light is also not on when the errors trigger. 

After homing, I did the `PROBE_ACCURACY` command and immediately received the following error: `Probe triggered prior to movement`. It was homed and the light was on when I ran the command so that condition makes sense...I'm not sure why that would have thrown an error though.

Jookia, 2024-11-15:
Ideally the probe should move upwards before probing to counteract this issue, which is what sample_retract_dist should do... Oops! That's not what it does.

You want to put this in your [safe_z_home] section: ```z_hop: 10``` (I just picked 10, try a value bigger if it doesn't work)

mminella, 2024-11-15:
No luck. When I put that in, it homed fine but then during the height map, I still get a `Probe triggered prior to movement` error. I increased the value to 20 and I got a `Endstop z still triggered after retract` during homing. I'm not sure how big you want that value to be...

Jookia, 2024-11-15:
> [Tracking image][https://github.com/notifications/beacon/AABNHO4WHZ7IR7MUMPUCYCL2AX5LBA5CNFSM6AAAAABRWR6GXCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUTYFG4M.gif]

Could you describe a bit more what the printer is doing? If you could provide a video that would be even better! It's hard to know exactly what's going on by just the errors as they don't show the entire picture.

Jookia, 2024-11-15:
Could you try both z hop and safe min retract distances?

mminella, 2024-11-15:
I have uploaded a video demonstrating the current state of what I'm trying here: https://youtu.be/i-T0G3fVJfU

Sorry, I recorded that video before seeing your last note. Do you know what the field name for safe min retract distance is? I can't find it in the docs. I'll give it a try once you let me know. Thanks again for your help!

Jookia, 2024-11-15:
I'm on my phone currently so I don't have the exact name but it was the previous setting we tried. You may need both of the settings to have the tool head move out of the way when homing and when probing.

I will check out the video later today, I'm currently in bed on my phone instead of sleeping. But at least that means I can reply to emails. :)

Jookia, 2024-11-16:
Okay, I had a look at your video, thank you very much for going so in depth.

This is very interesting behaviour. My mental picture of what was happening is that the after homing and probing the printer would go all the way down to the bed then not lift up high enough. Has this happened before? If so, I think the z_hop is the solution for that specific problem.

However in the video it seems like the printer doesn't move down very far at all. Maybe the reason why it is stopping is because the z controller thinks it's hit something, Could you try setting 'driver_SGTHRS: 100' in stepper_z? You might also want to set the run_current to 1 to see if that helps, I'm assuming you have both steppers running off the same controller so it might need a bit extra power.

mminella, 2024-11-18:
Thanks for the suggestions. I gave them a try and resulted in the same issue `Endstop z still triggered after retract`. I think (but I'm not sure) that I had tried various `driver_SGTHRS` values on the z before but none had a material impact (and I had assumed that was due to the fact that the Z was going to defer to the level sensor). I could be wrong though. I tried the 100 in the video, obviously I had it at 0 before, and I also tried 255...all the same results.

As for not lifting high enough, that hasn't been a consistent behavior. I wish it was because I agree that the solution would seem pretty straight foward.

I recorded another video in case it will help. Any help you can provide is very appreciated! https://youtu.be/SIWQbA9KIlw

Jookia, 2024-11-18:
Hmm, I think this is narrowing it down to a probing issue. Could you confirm that manually triggering the probe works, such as by putting some metal under the probe and making it light up? If so, does Klipper report any changes in the endstops section?

You may want to try these different values for the probe pin: PB8, ^PB8, !PB8, !^PB8 (or ^!PB8, I don't remember the order)

mminella, 2024-11-19:
I checked the probe and based on the light and state Klipper shows, you would think the probe itself works. That is captured in the video linked below.

One of the things I'm struggling with is the inconsistency of behavior. The pin variations is a good example of that. You had seen in videos I've shared that I couldn't even home correctly with the current configuration. Well you'll see in the video I was able to home and get half way through a height map before it failed. Today I was going to retry the various pin configuration options you suggested because I didn't do a full power cycle before each test to validate a clean and consistent starting point for each test. When  I did the first one again (PB8), I was able to home fine, I got a successful height map, but then when I tried to home doing the height map, I got the error `Endstop z still triggered after retract`. 

Sometimes the hardware and software seem to match up (the light goes on when it gets to the right point and the printer responds as expected) others it throws an error claiming that the Endstop z has been triggered in a place that is not expected (either after retract or prior to movement).

This video illustrates the above. You've already seen a video of the configuration using the PB8 pin in use and it wouldn't even home. This video shows that it will home and get half way through a height map before erroring out with a configuration we have tested before that wouldn't even home (for the record, as mentioned above, today I was able to home and do a height map with no issues...but homing afterwards failed). Let me know your thoughts.

https://youtu.be/WNh57HC9eY4

Jookia, 2024-11-19:
Hmm. This is tricky and points to some kind of physical issue. I would be interested if you could re-flash the original firmware and verify that that works.

Could you try swapping out the USB serial cable for a better quality one? You may need to buy one online. This is often the first source of problems with Klipper, it could be that some data is being corrupted due to noise.

If that doesn't work then checking connections on parts of the printer would be next to see that the cables are all undamaged and plugged in to their ports correctly.

I would also check the power connectors on the motherboard, if you haven't ferruled them you should do that when you get the time.

mminella, 2024-11-21:
I did reflash the original Kobra firmware (v 2.8.2 from here: https://store.anycubic.com/pages/firmware-software). I also switched my raspberry pi to Octoprint. I was able to home and level just fine multiple times. I actually played with the printer for a while last night with no real issues. Full disclosure, I did have to adjust the level sensor a tiny bit today as leveling in two spots with the native firmware was hitting the bed but it still completed fine. Once I made that adjustment, the printer has been solid again.

So at this point, I'm not 100% sure what to do. While I'd love to get Klipper running to try some things (like input shaping), the software when using Klipper acts like there are hardware errors which just do not appear when using the native firmware and Octoprint.

Jookia, 2024-11-21:
Okay, please keep me updated if you do further testing.

closed 2024-11-21 by Jookia