printers-for-people/catboat #52: Confirm Kobra 2022 working

gsohler, 2025-02-27:
Hi Catboat Team

Generally i can confirm that it works for my Konra2022 printer and i have printed my pi4 enclosure with it.

However. If repeatedly restarted the print DOES NOT work
The print works like
* bed heats up
* hotend heats up
* printer waits at least one minute, before i cancel

to get print working, need to push "Emergency Exit"  and reconnect

Happy to help debugging

And yes, when interested i can send the gcode file to look up the head up gcode command, but its not available at the moment

Jookia, 2025-02-27:
Hi there,

I'm glad to hear you managed to print an enclosure, nice.

Can you share your printer.cfg and klippy.log?

gsohler, 2025-02-27:
Hi , please  find enclosed the requested data,

[klippy.log](https://github.com/user-attachments/files/19015224/klippy.log)

[printer.txt](https://github.com/user-attachments/files/19015242/printer.txt)

Jookia, 2025-02-28:
These say the printer isn't connected to klipper. Is there an older klippy log from the error you describe?

Jookia, 2025-09-04:
I'm closing this due to lack of feedback. Please tell me if you need it re-opened.