printers-for-people/catboat #54: Not able to Access the klipper master File and catboat File to make the repository

monsti-cmd, 2025-06-02:
Hey i have the Problem that as well for klipper master and for the catboat File that are mentioned in the Installation Guide with kiuah are Not found anymore is there a workaround? I appreciate any Help

Jookia, 2025-06-02:
Could you please write the steps you've taken and show the error? What file is missing?

Jookia, 2025-08-25:
Thanks for the note. I'll try and fix this during the week.

mixednerdintx, 2025-09-04:
> Hi again people, I just went and set up KIAUH v6 using Catboat. I've put
> instructions here:
> https://github.com/printers-for-people/catboat/wiki/KIAUH-Install I
> didn't have to touch any kiauh.cfg, so maybe this was a KIAUH bug?

Looks like they beat you to the punch lol.

https://github.com/dw-0/kiauh/pull/718

Jookia, 2025-09-05:
Nice! I'll close this for now. Please ask me to re-open if this is still an issue. :)

closed 2025-09-05 by Jookia